OpenIntelWireless/IntelBluetoothFirmware #476: Bluetooth Controller Address is NULL | MacOS Sonoma |

brrandrade, 2024-03-06:
### Have you read the docs?

No

### macOS Version

Version 14.3.1 (23D60)

### Kext Version

v2.4.0

### Wireless Adapter Model and USB Product ID

INTEL  Vendor ID 0x8087 | Product ID 0x0029

### Description

- USB Port Mapping is done
- Adapter shows under System Info -> USB
- Bluetooth Controller is NULL under System Info -> Bluetooth
- Kexts are loading
- Bluetooth switch can be switched on and off in Settings and in Control Center
- No devices are found 
- Kexts:
- [image]

- boot-args:  keepsyms=1 -btlfxallowanyaddr -wegnoigpu agdpmod=pikera alcid=23 -dbgenhdbg
 
- kextstat output:
xxx@iMac-Pro ~ % kextstat | grep -E "BTPatcher|IntelBluetoothFirmware|BlueToolFixup|DebugEnhancer"
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   59    0 0xffffff8004c87000 0xa000     0xa000     as.lvs1974.DebugEnhancer (1.0.9) A436CC6E-646F-3921-AD16-5151136BC7E2 <58 9 7 6 3 2 1>
   60    0 0xffffff8004c91000 0xf000     0xf000     com.zxystd.IntelBTPatcher (2.4.0) 06D53ACB-3381-3B25-BBA8-403841B1AE25 <58 30 9 7 6 3 2 1>
   61    0 0xffffff800579d000 0xa000     0xa000     as.acidanthera.BlueToolFixup (2.6.8) F79E90EC-E91C-3D93-83D1-C0C5C9206285 <58 9 7 6 3 2 1>
  105    0 0xffffff8004ca0000 0xafd000   0xafd000   com.zxystd.IntelBluetoothFirmware (2.4.0) F9DD55B4-27C6-39D5-AAFC-C599DB3DEBCA <30 7 6 3>

USB Mapping in Hackintool:
[image]

IOREg
[image]

### Info in System Report - Bluetooth

```shell
Hardware -> Bluetooth
 Bluetooth Controller:
  Address:	NULL
  State:	On
  Discoverable:	On
  Vendor ID:	0x004C (Apple)

Hardware -> USB
USB32Bus:

  Host Controller Driver:	AppleUSBXHCIPCI
  PCI Device ID:	0x7ae0 
  PCI Revision ID:	0x0011 
  PCI Vendor ID:	0x8086 

Bluetooth USB Host Controller:

  Product ID:	0x0029
  Vendor ID:	0x8087  (Intel Corporation)
  Version:	0.01
  Speed:	Up to 12 Mb/s
  Location ID:	0x00c00000 / 7
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0
  Built-In:	Yes
```

### Relevant log output

```shell
sudo dmesg | grep -i IntelBluetooth
[    0.988858]: IOUSBHostDevice@00c00000: IOUSBHostDevice::setConfigurationGated: IntelBluetoothFirmware selected configuration 1
```

brrandrade, 2024-03-06:
IOREG FILE: [iMac Pro.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/14516033/iMac.Pro.zip)

zxystd, 2024-03-09:
The log is incomplete, the actual keyword is 'IntelFirmware' but not 'IntelBluetooth', please collect again.

derpuma, 2024-03-10:
I have the same problem. The IntelBluetoothFirmware.kext driver is loaded with ibt-0041-0041.sfi.
I see BT under USB Hardware, but the Controller under Bluetooth show ist with [Null]-
[image]
[image]
[image]
[image]

Kextstat shows that all relevant drivers are loaded:
[image]

zxystd, 2024-03-11:
@brrandrade @derpuma From your logs, we can see the IntelBluetoothFirmware firmware driver is working fine, so your issue is about Hackintosh, that's out of our control, please see other way for help. Thanks.

zxystd, 2024-03-11:
You can upload a log of `bluetoothd`, maybe I can help you locate what's the problem.
```
log show --last 5m | grep bluetoothd > ~/desktop/a.txt
```

closed 2024-03-11 by zxystd

derpuma, 2024-03-11:
Of course! It would be be great if you could help. I am not on a hackintosh. My macOS runs in a KVM, but should act like a hacky!

As long as the BT shows up as a USB Device, it should also be loaded as a Bluetooth device.
Not sure were the quirk is here...

Here is the logfile:
[a.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/14555545/a.txt)

brrandrade, 2024-03-11:
I am also running in KVM, it was ok even on Ventura, I have collected the log and toggled off and on the bluetooth to populate more info.
[a.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/14557829/a.txt)

derpuma, 2024-03-12:
> I am also running in KVM, it was ok even on Ventura, I have collected the log and toggled off and on the bluetooth to populate more info. [a.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/14557829/a.txt)

Hey @brrandrade. My bluetooth works now...

Please feel free to add this kext into your EFI/OC/Kext and add it to your config.plist:

[Bluetooth File Exchange.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/14576919/Bluetooth.File.Exchange.kext.zip)

And please post results.
See the results on my side now!

[image]

brrandrade, 2024-03-12:
Hi @derpuma, that is great that you got it working, I jus tried your kext, but didn't work for me.
My specs:
INTEL i9-12900K
GPU AMD Radeon RX 580 8 GB
Motherboard: Aorus Z690 PRO DDR5

Your kext is not loading here:
kextstat | grep bluetooth -i
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   50    0 0xffffff800236a000 0x37000    0x37000    com.apple.iokit.IOBluetoothFamily (9.0.0) 94011177-9DC3-353A-85CD-A0081589A3DB <49 38 17 9 8 7 6 3 1>
   98    0 0xffffff8003d16000 0xafd000   0xafd000   com.zxystd.IntelBluetoothFirmware (2.4.0) F9DD55B4-27C6-39D5-AAFC-C599DB3DEBCA <38 7 6 3>
   
   Thanks for the help.

derpuma, 2024-03-13:
Here you have my EFI Folder, with the thing I tried to get it to work.
To be honest, there are so many inputs I got. I am not 100% sure, what brought Bluettooth to life in the end.
Maybe you can look and find something that helps you too...

https://mega.nz/file/SVQmwALI#kZVztxxoQNbKNRBuhD-UH5lsbqLXg5K_iM5pwQzM58g

maxpicelli, 2024-03-13:
@brrandrade Bruno, tente usar a EFI do Derpuma, pois eu a modifiquei e refiz alguns parâmetros na plist e as minhas Kexts da intel. A plist dele estava faltando algumas modificações.  As Kexts que ele se refere são de um amigo que é programador. 

Try use @derpuma EFI, im modified some parameters, and use my Intel kexts. Look especially in SecureBootModel, was X86platform but in fact, Intel or Ryzen (or other processors) is better use **Disabled** on plist 

If you can try use those kexts to help your Bluetooth connexion, check GitHub developer page 

https://github.com/hnanoto/AirPort-Utility and here https://github.com/hnanoto/Bluetooth-File-Exchange

maxpicelli, 2024-03-13:
@brrandrade você pode acessar o Discord ou o Youtube de nosso amigo Henrique Nonato 

Veja no canal **_Hackintosh and Beyond_** do Youtube, Ali tem todos os links de acesso.

brrandrade, 2024-03-13:
Hi @derpuma, thank you for your EFI, it worked and I can't believe lol, I was hopeless with that.

It worked even without using usbport kext and with the XHCIPortLimit enabled.

@maxpicelli, I didn't use your kexts, do you think it will bring something for my intel, or leave as it is?
I got the channels in youtube and discord, thanks for that.

I just had one Problem, I can't boot with boot-args= -v, it gets stuck as showed in the picture below, without it, boot normally, do you have any idea what can I do to enable debug without get stuck?

[image]

derpuma, 2024-03-14:
@brrandrade I guess, as we are running macOS in a KVM, it is possible, that you run your macOS also with my AMD kext from my EFI folder. The KVM emulates the hardware, the kext pretend to be... I bet it doesn't make any difference to have some kext or not.
As I do not know, what exactly made the difference for BT to be detected as the right Controller, I am not sure at all. I will investigate more over the weekend and try to specify which enabled it at the end. For now I am happy, thanks to @maxpicelli who gave me a helping hand at insanelymac forum.

maxpicelli, 2024-03-19:
Maybe you should try install in normally without KVM. After that, use same EFI to try boot on KVM. Need extract DMAR table, with SSDTTime or similar and so enable VT-d.

brrandrade, 2024-03-21:
I did extracting it from linux and windows, but both in KVM got stuck at boot, I need to try with a mac installation, it might generate different files, but for now, no extra disk to try it. but the goal here was reached with the bluetooth working, thanks again :).

Ulytas, 2024-11-04:
@derpuma Hi and thank you very much for sharing your files. I copied the config.plist and all the kexts. It works on a KVM Sonoma 14.7 made with OSX-PROXMOX.

brrandrade, 2024-12-23:
Hi @maxpicelli @derpuma , did you manage to make it to work on Sequoia?
Best regards

maxpicelli, 2024-12-23:
Hi @derpuma i have been used Broadcom Wi-Fi and Bluetooth with OCLP patcher and EFI configurations required to make it works. God bless you ! The coder not update anymore. Maybe tou should try change too.

brrandrade, 2024-12-23:
I was able to make the wifi to work using the steps in https://elitemacx86.com/threads/how-to-fix-intel-wifi-and-bluetooth-on-macos-sequoia-and-later.2066/, but the bluetooth, probably because of something missing in the virtualization, is not working.
Happy new year and God Bless you all too, thanks for keeping it alive.

maxpicelli, 2024-12-24:
Thanks @brrandrade sorry that i forgot mention you. I hope you can fix that. I know the owner of ElitemacX86, is a friend my Amit Sinhá, Indian Coder. Good place. Thanks you

CozyDust, 2025-06-01:
> > > I was able to make the wifi to work using the steps in https://elitemacx86.com/threads/how-to-fix-intel-wifi-and-bluetooth-on-macos-sequoia-and-later.2066/, but the bluetooth, probably because of something missing in the virtualization, is not working. Happy new year and God Bless you all too, thanks for keeping it alive.
> > 
> > 
> > Thanks, I have the same bluetooth problem and I finally fix it using the steps in the tutorial.
> 
> Hi, that is great, did you use it in KVM? could you share your EFI?

Not in KVM, I installed Hackintosh directly on my laptop. My EFI is still being tested with various modifications, so it doesn’t have much reference value. My issue was that I didn’t set up NVRAM properly. After configuring bluetoothExternalDongleFailed and bluetoothInternalControllerInfo according to the tutorial, Bluetooth started working normally. You might want to check these settings.

lwlf, 2025-06-15:
> I was able to make the wifi to work using the steps in https://elitemacx86.com/threads/how-to-fix-intel-wifi-and-bluetooth-on-macos-sequoia-and-later.2066/, but the bluetooth, probably because of [...]

I just solved the problem of Sequoia driving Intel bluetooth under kvm in these days, you can try to install VMHide.kext and put it after Lilu.kext.

PS: May I ask if your Intel wifi has the problem of losing wifi device after restarting Sequoia VM?

brrandrade, 2025-06-16:
> I just solved the problem of Sequoia driving Intel bluetooth under kvm in these days, you can try to install VMHide.kext and put it after Lilu.kext.
>
> PS: May I ask if your Intel wifi has the problem [...]

Hi, nice kext! I didn’t know about it. Unfortunately, my GPU (AMD Radeon RX 580 8 GB) isn’t working with the latest Sequoia version. I can't get past the Apple logo when using KVM.

I need to figure out what's going on. If I manage to get it working, I'll check the Wi-Fi and let you know. By the way, which Sequoia version are you using?

lwlf, 2025-07-02:
> Hi, thanks, I tried and tried and no luck, so, decided to sell and bought RX 6650 XT, I am having problems to make it to work, even spoofing it, but no luck, could you please share your EFI, args from VM and the extra kernel parameters from host as pci.... to make the AMD to work properly. thanks again, it will help me a lot.

Hi, here is my config file, you can refer to the following but please do not copy it directly.

`cat /etc/pve/qemu-server/13000.conf`:

```conf
agent: 1
args: -device isa-applesmc,osk="ourhardworkbythesewordsguardedpleasedontsteal(c)AppleComputerInc" -smbios type=2 -device usb-kbd,bus=ehci.0,port=2 -global nec-usb-xhci.msi=off -global ICH9-LPC.acpi-pci-hotplug-with-bridge-support=off -cpu Haswell-noTSX,vendor=GenuineIntel,+invtsc,+hypervisor,kvm=on,vmware-cpuid-freq=on
balloon: 0
bios: ovmf
boot: order=virtio0;net0
cores: 8
cpu: host
efidisk0: pve-01-nvme-01-4tb:13000/vm-13000-disk-0.raw,efitype=4m,size=528K
hostpci0: mapping=PVE-01_NIC-01_Realtek-RTL8215,pcie=1
hostpci1: mapping=PVE-01_WNIC-01_Intel-AX210,pcie=1
hostpci2: mapping=PVE-01_GPU-02_AMD-RX6600XT_with-Audio,pcie=1
localtime: 0
machine: pc-q35-9.2+pve1
memory: 32768
meta: creation-qemu=9.2.0,ctime=1750336277
name: macOS
net0: virtio=BC:24:11:D1:9A:08,bridge=vmbr0,firewall=1
numa: 1
ostype: win11
parent: Snapshot_Release_00-03
protection: 1
scsihw: virtio-scsi-single
smbios1: uuid=7709fea0-36e6-4e8d-8229-6e0527166272
sockets: 1
tags: macOS;Sequoia
usb0: mapping=PVE-01_Bluetooth-01_Intel-AX210,usb3=1
vga: none
virtio0: pve-01-nvme-01-4tb:13000/vm-13000-disk-1.raw,cache=unsafe,discard=on,iothread=1,size=1T
```

`cat /etc/modprobe.d/vfio.conf`:

```conf
options vfio-pci ids=8086:2725
```

`cat /etc/modprobe.d/iommu_unsafe_interrupts.conf `:

```conf
options vfio_iommu_type1 allow_unsafe_interrupts=1
```

`cat /etc/modules-load.d/vfio.conf `:

```conf
vfio
vfio_iommu_type1
vfio_pci
```

`cat /etc/default/grub.d/iommu.cfg `:

```conf
GRUB_CMDLINE_LINUX_DEFAULT="quiet iommu=pt pcie_acs_override=downstream,multifunction biosdevname=0 clocksource=tsc tsc=reliable mitigations=off pcie_aspm=off"
```

EFI files:
> I can't upload the attachment on my network, I'll share it with you later using Google Cloud Drive.

brrandrade, 2025-07-03:
> > Hi, thanks, I tried and tried and no luck, so, decided to sell and bought RX 6650 XT, I am having problems to make it to work, even spoofing it, but no luck, could you please share your EFI, args from VM and the extra kernel parameters from host as pci.... to make the AMD to work properly. thanks again, it will help me a lot.
> 
> EFI File: https://drive.google.com/file/d/1HBRxQvUfBu1rNTYtUcqQ7oaAuf93rNKx/view?usp=sharing

Hey, thanks a lot! I've done a lot of testing, and now I have Sequoia working with Bluetooth. The culprit turned out to be that I was trying to install it on a virtual IDE disk, which caused it to freeze only when injecting the GPU — really weird.

I also noticed something interesting that might help someone here: to see verbose mode with the GPU without freezing, you need to disable “All Functions,” “Primary GPU,” and “PCI-Express.” After doing that, I was able to see the loading bar and verbose output working properly.

As for Wi-Fi, I tried a lot. The most I could achieve was turning it on, but it wasn't listing any networks. Other attempts caused the system to freeze during boot. Sorry I couldn't help more with that.

One more difference from your EFI: I had to use iMac20,1. The iMac7,1 config kept freezing for me, and I'm not sure why.

The good news is that I tried booting the Tahoe Beta installer with the same configuration, and it worked — though I didn't let it finish due to lack of time.

Thanks again!

lwlf, 2025-07-04:
> Hey, thanks a lot! I've done a lot of testing, and now I have Sequoia working with Bluetooth. The culprit turned out to be that I was trying to install it on a virtual IDE disk, which caused it to [...]

Congratulations and happy for you. 

Are you using a Broadcom wireless card? I've tried it and it only works with open (i.e. no password) wifi under the VM, connecting to encrypted wifi causes a crash and reboot. I haven't found a suitable solution on my end. This should be a Linux kernel issue, it is possible to drive and connect to encrypted wifi on 5.15 kernel. pve 8 uses 6.0+ kernel.

I have searched and tried for a long time, but no solution. Now using Intel AX210, it works with encrypted wifi, it's just that I need to manually disconnect the wifi connection before shutting down and rebooting, instead of just shutting down the wifi device.

The macos Tahoe I've also tried, the current EFI doesn't seem to be able to drive the full security, only up to the configure system session, and will not be able to use it going forward. It may be because I'm using an AMD CPU, or it may work fine with Intel.

Good luck.