OpenIntelWireless/IntelBluetoothFirmware #478: Sonoma 14.4 will auto reboot

dannyCaiHaoming, 2024-03-09:
### Have you read the docs?

No

### macOS Version

macOS 14.4

### Kext Version

V2.2.0

### Wireless Adapter Model and USB Product ID

Intel AX 210

### Description

I upgraded sonoma 14.4 today, and when I entered the password to access the main page, the system automatically restarted, and the same is true for multiple attempts

### Info in System Report - Bluetooth

```shell
The system cannot be accessed, so the command line cannot be used
```

### Relevant log output

```shell
The system cannot be accessed, so the command line cannot be used
```

zxystd, 2024-03-09:
BT is working fine on 14.4, please make sure you are using Sonoma 14.4 specific version of AirportItlwm.

dannyCaiHaoming, 2024-03-09:
> BT is working fine on 14.4, please make sure you are using Sonoma 14.4 specific version of AirportItlwm.

thx.It worked when I used AirportItlwm v2.3.0-alpha