OpenIntelWireless/IntelBluetoothFirmware #479: Bluetooth broken on macOS Sonoma 14.4 Not working after updating it from 14.3.1 ( last known working version )

watarikai96, 2024-03-11:
### Have you read the docs?

Yes

### macOS Version

macOS 14.4

### Kext Version

v2.4.0

### Wireless Adapter Model and USB Product ID

Intel AX211 / AX210 / AX201 / AX200

### Description

The bluetooth was working on Sonoma 14.3.1 after updating to 14.4 it stopped working. 
I tried everything still not working.
[image]

### Info in System Report - Bluetooth

```shell
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               bluetoothd [4445]
Path:                  /usr/sbin/bluetoothd
Identifier:            bluetoothd
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               0

Date/Time:             2024-03-11 16:38:30.4533 +0900
OS Version:            macOS 14.4 (23E214)
Report Version:        12
Anonymous UUID:        DD119664-D9CB-A825-CFE3-6016320F690F

Time Awake Since Boot: 3100 seconds

System Integrity Protection: enabled

Crashed Thread:        1

Exception Type:        EXC_GUARD
Exception Codes:       GUARD_TYPE_USER
Exception Codes:       0x6000000000000012, 0x0000000000000002

Termination Reason:    Namespace LIBSYSTEM, Code 2 Application Triggered Fault

Thread 0:
0   libsystem_kernel.dylib        	    0x7ff80b03a3ce mach_msg2_trap + 10
1   libsystem_kernel.dylib        	    0x7ff80b048c88 mach_msg2_internal + 84
2   libsystem_kernel.dylib        	    0x7ff80b041510 mach_msg_overwrite + 653
3   libsystem_kernel.dylib        	    0x7ff80b03a6bd mach_msg + 19
4   CoreFoundation                	    0x7ff80b156be5 __CFRunLoopServiceMachPort + 143
5   CoreFoundation                	    0x7ff80b155655 __CFRunLoopRun + 1371
6   CoreFoundation                	    0x7ff80b154b32 CFRunLoopRunSpecific + 557
7   CoreFoundation                	    0x7ff80b1d05be CFRunLoopRun + 40
8   bluetoothd                    	       0x105c6710a 0x105c38000 + 192778
9   dyld                          	    0x7ff80acee366 start + 1942

Thread 1 Crashed:
0   libsystem_kernel.dylib        	    0x7ff80b045aca os_fault_with_payload + 10
1   bluetoothd                    	       0x105d3bff3 0x105c38000 + 1064947
2   bluetoothd                    	       0x105ce1ea2 0x105c38000 + 695970
[... 503 lines cut ...]
    "uuid" : "5a006814-83f4-3c80-ab1f-528cdd08c0ff",
    "path" : "\/usr\/lib\/system\/libsystem_pthread.dylib",
    "name" : "libsystem_pthread.dylib"
  }
],
  "sharedCache" : {
  "base" : 140703309225984,
  "size" : 25769803776,
  "uuid" : "1133465d-2127-33d2-9460-441064d34f11"
},
  "vmSummary" : "ReadOnly portion of Libraries: Total=678.0M resident=0K(0%) swapped_out_or_unallocated=678.0M(100%)\nWritable regions: Total=521.6M written=0K(0%) resident=0K(0%) swapped_out=0K(0%) unallocated=521.6M(100%)\n\n                                   VIRTUAL   REGION \nREGION TYPE                           SIZE    COUNT (non-coalesced) \n===========                        =======  ======= \nActivity Tracing                      256K        1 \nDispatch continuations              128.0M        1 \nDispatch continuations (reserved)    64.0M        1         reserved VM address space (unallocated)\nIOKit                                  16K        2 \nKernel Alloc Once                       8K        1 \nMALLOC                              318.3M       55 \nMALLOC guard page                      48K       12 \nSQLite page cache                     384K        3 \nSTACK GUARD                          56.0M        5 \nStack                                10.0M        6 \nStack (reserved)                      520K        1         reserved VM address space (unallocated)\nStack Guard                             4K        1 \nVM_ALLOCATE                            28K        3 \n__CTF                                  824        1 \n__DATA                               26.8M      516 \n__DATA_CONST                         32.0M      344 \n__DATA_DIRTY                         1371K      191 \n__FONT_DATA                           2352        1 \n__INFO_FILTER                            8        1 \n__LINKEDIT                          180.6M        4 \n__OBJC_RO                            71.7M        1 \n__OBJC_RW                            2196K        2 \n__TEXT                              497.4M      531 \ndyld private memory                   260K        2 \nmapped file                          31.6M        8 \nshared memory                         596K        8 \n===========                        =======  ======= \nTOTAL                                 1.4G     1702 \nTOTAL, minus reserved VM space        1.3G     1702 \n",
  "legacyInfo" : {
  "threadTriggered" : {

  }
},
  "logWritingSignature" : "33dfccb6252eb59b0f18855fad94a65abd61a9c3"
}
```

### Kernel Panic Logs

```shell
None
```

watarikai96, 2024-03-11:
The bluetooth was perfectly working fine just a version before on 14.3.1 maybe the BluetoothFixup.kext needs some updating?

zxystd, 2024-03-12:
@watarikai96 Others using the same kext are working fine, and if you are asking BluetoothFixup.kext 's problem, please turn to [BrcmPatchRAM](https://github.com/acidanthera/BrcmPatchRAM) repo, not here.

closed 2024-03-12 by zxystd

watarikai96, 2024-03-12:
NVM it's fixed! Thank you!

YiPhuongTrinh, 2024-03-27:
> NVM it's fixed! Thank you!

How can you fix it? I have same problem too. 14.3 everything is okay but after 14.4 Bluetooth not work anymore.