OpenIntelWireless/IntelBluetoothFirmware #484: Bluetooth stop working after wake up from sleep

jiatao99, 2024-05-05:
### Have you read the docs?

Yes

### macOS Version

macOS 13

### Kext Version

v2.4.0

### Wireless Adapter Model and USB Product ID

Intel AX200

### Description

After wakeup from sleep, wifi works but bluetooth is gone. Have to reboot

[image]

When first log into mac, bluetooth was working fine. Show as "THIRD_PARTY DONGLE"

Tested bluetoothInternalControllerInfo and bluetoothExternalDongleFailed flag. No effect

### Info in System Report - Bluetooth

```shell
...
```

### Relevant log output

```shell
Nothing shows up after sleep
```

zhiyuan-mojie, 2024-05-29:
macos 14.4.1 have the same problem.

zxystd, 2024-05-29:
it should be fix by yourself, not relate to this kext.

closed 2024-05-29 by zxystd