OpenIntelWireless/IntelBluetoothFirmware #486: Kernel panik when activating IntelBTPatcher.kext on SEQUOIA

anonymous-writer, 2024-06-13:
### Have you read the docs?

Yes

### macOS Version

macOS 15

### Kext Version

v2.4.0

### Wireless Adapter Model and USB Product ID

Intel AX-210

### Description

Kernel panik when activating IntelBTPatcher.kext. Panic attached.

### Info in System Report - Bluetooth

```shell
Unfortunately no information because of the kernel panic.
```

### Relevant log output

```shell
Unfortunately no information because of the kernel panic.
```

### Kernel Panic Logs

```shell
panic(cpu 7 caller 0xffffff801a90e4c4): IOGMD: not wired for the IODMACommand com.apple.IOMemoryDescriptor.cpp:3246
Panicked task 0xffffff8b69166870: 15 threads: pid 142: bluetoothd
Backtrace (CPU 7), panicked thread: 0xffffff903608f0c8, Frame : Return Address
0xffffffd101223240 : 0xffffff801a1644e1 
0xffffffd101223290 : 0xffffff801a2cfefc 
0xffffffd1012232d0 : 0xffffff801a2bfbad 
0xffffffd101223370 : 0xffffff801a0fa971 
0xffffffd101223390 : 0xffffff801a1647d7 
0xffffffd101223490 : 0xffffff801a163e72 
0xffffffd101223500 : 0xffffff801a9d3d58 
0xffffffd1012235f0 : 0xffffff801a90e4c4 
0xffffffd101223660 : 0xffffff801a90e791 
0xffffffd101223740 : 0xffffff801a90a44c 
0xffffffd1012237a0 : 0xffffff801e0b599b 
0xffffffd101223900 : 0xffffff801ca18f97 
0xffffffd101223930 : 0xffffff801ca34363 
0xffffffd101223990 : 0xffffff801ca34254 
0xffffffd101223a10 : 0xffffff801a93b7a6 
0xffffffd101223a40 : 0xffffff801a93bfed 
0xffffffd101223ba0 : 0xffffff801a2776ef 
0xffffffd101223d00 : 0xffffff801a1414b8 
0xffffffd101223da0 : 0xffffff801a154c45 
0xffffffd101223df0 : 0xffffff801a155334 
0xffffffd101223ee0 : 0xffffff801a2a44aa 
0xffffffd101223fa0 : 0xffffff801a0fadd6 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOUSBHostFamily(1.2)[A012230A-389F-3468-889F-5F8733B17D45]com.apple.0xffffff801c9db000-com.apple.driver.0xffffff801ca4ffff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[27641E32-63A0-37ED-B1F7-326703B80BA4]com.apple.0xffffff801b52c000-com.apple.driver.0xffffff801b52ffff
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[81226F90-21C1-3DDD-8374-49170AB2757B]com.apple.0xffffff801b83b000-com.apple.driver.0xffffff801b873fff
            dependency: com.apple.driver.AppleSMC(3.1.9)[B44C843C-D46E-307A-8968-3EEFC6B9E6F2]com.apple.0xffffff801b92b000-com.apple.driver.0xffffff801b944fff
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[8509402D-83BE-34CF-B822-3FBA13724ED0]com.apple.0xffffff801bb44000-com.apple.driver.0xffffff801bb47fff
            dependency: com.apple.driver.AppleUSBHostMergeProperties(1.2)[E075EB10-9189-3B10-BF6A-734720C92E37]com.apple.0xffffff801caae000-com.apple.driver.0xffffff801caaefff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[5CCDA23F-974B-36FE-991B-F492F97C7C90]com.apple.0xffffff801c147000-com.apple.driver.0xffffff801c148fff
         com.zxystd.IntelBTPatcher(2.5)[0C32ABE2-93E3-30C9-9B28-2A220C5FF100]com.apple.0xffffff801e0b4000-com.apple.driver.0xffffff801e0bcfff
            dependency: as.vit9696.Lilu(1.6.8)[B010EDEA-1824-3683-97AC-9BAE10C3D851]com.apple.0xffffff801dc32000-com.apple.driver.0xffffff801dc5dfff
[... 183 lines cut ...]
com.apple.driver.AppleBusPowerController	1.0
com.apple.iokit.IOSCSIBlockCommandsDevice	498
com.apple.iokit.IOSCSIArchitectureModelFamily	498
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleEffaceableStorage	1.0
com.apple.iokit.IOStorageFamily	2.1
com.apple.driver.AppleCyrus	1
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.security.AppleImage4	7.0.0
com.apple.kext.CoreTrust	1
com.apple.driver.AppleCredentialManager	1.0
com.apple.iokit.CoreAnalyticsFamily	1
com.apple.driver.AppleSEPManager	1.0.1
com.apple.driver.IOSlaveProcessor	1
com.apple.driver.AppleACPIPlatform	6.1
com.apple.driver.AppleSMC	3.1.9
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.driver.watchdog	1
com.apple.kec.pthread	1
com.apple.kec.XrtHostedXnu	1
com.apple.kec.Libm	1
com.apple.kec.Compression	1.0
com.apple.kec.corecrypto	14.0
```

johnpapad24, 2024-06-13:
problem confirmed but I am personally fine with the bluetooth on sequoia without the use of this kext.

(intel 8260)

I believe that just disabling this kext will do the trick

anonymous-writer, 2024-06-13:
Sorry, but BT 5.3 need this Kext.

johnpapad24, 2024-06-13:
try this configuration:

IntelBluetoothFirmware.kext
BlueToolFixup.kext
BrcmPatchRAM2.kext
BrcmPatchRAM3.kext

tunglamvghy, 2024-06-13:
I have the same issue. Do you know the solution?

anonymous-writer, 2024-06-13:
Thanks for the answers. 
But it would be good if the answers here were from Users with Bluetooth module higher than 5.2.
Only these need the Kext and there is no other known solution Monterey.
I have adjusted the headline to Bluetooth higher 5.2.

anonymous-writer, 2024-06-14:
There ist clearly written "Do not use it on Monterey". Which certainly means beginning from Monterey.
Apart from that, I have tried Kext IntelBluetoothFirmware. SEQUOIA stop start with this kext.

It would be really great if only those who have experience with Intel and Bluetooth higher than 5.1 answered here.

Lorys89, 2024-06-17:
[BlueToolFixup.kext.zip](https://github.com/user-attachments/files/15875858/BlueToolFixup.kext.zip)
[IntelBluetoothFirmware-2.5.0-RELEASE.zip](https://github.com/user-attachments/files/15875860/IntelBluetoothFirmware-2.5.0-RELEASE.zip)

but add nvram setting

bluetoothExternalDongleFailed DATA 00

bluetoothInternalControllerInfo DATA 0000000000000000000000000000

anonymous-writer, 2024-06-19:
@Lorys89 ,
The kexts you posted work perfectly. Thank you very much for that.

When created from your source code, the kext doesn't work for me. For whatever reason?
https://github.com/Lorys89/IntelBluetoothFirmware

Lorys89, 2024-06-19:
> @Lorys89 , The kexts you posted work perfectly. Thank you very much for that.
> 
> I don't need this: bluetoothExternalDongleFailed DATA 00 bluetoothInternalControllerInfo DATA 0000000000000000000000000000
> 
> I only use the start parameter: -btlfxallowanyaddr If I really need this parameter i have not tested.
> 
> When created from your source code, the kext doesn't work for me. For whatever reason. https://github.com/Lorys89/IntelBluetoothFirmware

Use this 
https://github.com/Lorys89/IntelBluetoothFirmware/actions/runs/9553221087/artifacts/1609892363

xfwdrev, 2024-06-25:
> Use this https://github.com/Lorys89/IntelBluetoothFirmware/actions/runs/9553221087/artifacts/1609892363

This one worked. But apparently it doesn't work again on Beta 2.

nguyentdat23, 2024-06-26:
> @Lorys89 , The kexts you posted work perfectly. Thank you very much for that.
>
> I don't need this: bluetoothExternalDongleFailed DATA 00 bluetoothInternalControllerInfo DATA [...]

I follow your instruction and bluetooth can be turn on but failed to discover

nguyentdat23, 2024-06-26:
> [BlueToolFixup.kext.zip](https://github.com/user-attachments/files/15875858/BlueToolFixup.kext.zip) [...]

I follow your instruction and bluetooth can be turn on but failed to discover. Is there an order for sort these kext?

Lorys89, 2024-06-26:
Try this https://github.com/Lorys89/IntelBluetoothFirmware/releases/download/2.5.0/IntelBluetoothFirmware-2.5.0-RELEASE.zip

nguyentdat23, 2024-06-26:
> Try this https://github.com/Lorys89/IntelBluetoothFirmware/releases/download/2.5.0/IntelBluetoothFirmware-2.5.0-RELEASE.zip

Thanks so much mate! Now bluetooth works flawless in Sequoia Beta 2 with Intel AX201.

mick3lson, 2024-06-26:
> Thanks so much mate! Now bluetooth works flawless in Sequoia Beta 2 with Intel AX201.

I tried this kext but don't work on my Intel AX200, can you explain wich kext did you use and if nvram setting is mandatory.

xfwdrev, 2024-06-26:
> Try this https://github.com/Lorys89/IntelBluetoothFirmware/releases/download/2.5.0/IntelBluetoothFirmware-2.5.0-RELEASE.zip

Thanks. Works again on beta 2. Have to perform NVRAM Clear to make it work too.

nguyentdat23, 2024-06-29:
> > Thanks so much mate! Now bluetooth works flawless in Sequoia Beta 2 with Intel AX201.
> 
> I tried this kext but don't work on my Intel AX200, can you explain wich kext did you use and if nvram setting is mandatory.

I add these kext with order like the image below, IntelBluetoothFirmware, IntelBTPatcher and BlueToolFixup i got from Lorys89 and add two NVRAM like Lorys89's instruction
[image]
[image]

Daasin, 2024-07-06:
Is this newer than the dortania build list ver?

Daasin, 2024-07-15:
Worked on one device, 
Not on the other

> `panic(cpu 0 caller 0xffffff800610c7f4): IOGMD: not wired for the IODMACommand @IOMemoryDescriptor.cpp:3246
Panicked task 0xffffff9045d259b0: 181 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffff99de7b6b30, Frame : Return Address
0xffffffea16b2f530 : 0xffffff800595e641 
0xffffffea16b2f580 : 0xffffff8005acb14c 
0xffffffea16b2f5c0 : 0xffffff8005aba7cb 
0xffffffea16b2f680 : 0xffffff80058f4971 
0xffffffea16b2f6a0 : 0xffffff800595e937 
0xffffffea16b2f7a0 : 0xffffff800595dfd2 
0xffffffea16b2f810 : 0xffffff80061d47f8 
0xffffffea16b2f900 : 0xffffff800610c7f4 
0xffffffea16b2f970 : 0xffffff800610cac1 
0xffffffea16b2fa50 : 0xffffff800610881c 
0xffffffea16b2fab0 : 0xffffff800a28698e 
0xffffffea16b2fb00 : 0xffffff800823e3e9 
0xffffffea16b2fcf0 : 0xffffff8007356d3b 
0xffffffea16b2fef0 : 0xffffff80073568b4 
0xffffffea16b2ff20 : 0xffffff80060f800e 
0xffffffea16b2ff60 : 0xffffff80060f7646 
0xffffffea16b2ffa0 : 0xffffff80058f419e 
      Kernel Extensions in backtrace:
         com.apple.driver.usb.AppleUSBCommon(1.0)[275A45BD-9BA2-35B8-8DE4-619323D72692]@0xffffff8007354000->0xffffff8007357fff
         com.apple.iokit.IOUSBHostFamily(1.2)[C82923C2-8D24-30D5-8D62-55BD7F33E700]@0xffffff80081ec000->0xffffff8008260fff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[27641E32-63A0-37ED-B1F7-326703B80BA4]@0xffffff8006d2f000->0xffffff8006d32fff
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[CABC2D45-E2FF-38EA-916D-87EEEEF51006]@0xffffff8007045000->0xffffff800707dfff
            dependency: com.apple.driver.AppleSMC(3.1.9)[AE2D404B-52C3-3E5E-9E50-1CFE4C8CB5B2]@0xffffff8007136000->0xffffff800714ffff
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[275A45BD-9BA2-35B8-8DE4-619323D72692]@0xffffff8007354000->0xffffff8007357fff
            dependency: com.apple.driver.AppleUSBHostMergeProperties(1.2)[17B1956A-6399-3572-91C8-01671E9E18A9]@0xffffff80082bf000->0xffffff80082bffff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[5CCDA23F-974B-36FE-991B-F492F97C7C90]@0xffffff8007956000->0xffffff8007957fff
         com.zxystd.IntelBTPatcher(2.5)[8B35E4E1-C7C8-3963-A181-4190BBA01315]@0xffffff800a285000->0xffffff800a28efff
            dependency: as.vit9696.Lilu(1.6.7)[087C34D2-49F8-3FDA-8757-27E9425C9EE4]@0xffffff80094af000->0xffffff80094dafff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[C82923C2-8D24-30D5-8D62-55BD7F33E700]@0xffffff80081ec000->0xffffff8008260fff

 > 'Process name corresponding to current thread (0xffffff99de7b6b30): kernel_task
Boot args: -no_compat_check -lilubetaall -wegbeta -alcbeta -igfxblt igfxonln=1 revpatch=sbvmm -btlfxallowanyaddr'

> 'Mac OS version:
24A5289h
Kernel version:
Darwin Kernel Version 24.0.0: Mon Jul  1 21:56:48 PDT 2024; root:xnu-11215.0.132.501.1~1/RELEASE_X86_64
Kernel UUID: 038C92A2-763E-35BA-A79F-E1835BE4D67E
roots installed: 0
KernelCache slide: 0x0000000005600000
KernelCache base:  0xffffff8005800000
Kernel slide:      0x00000000056e4000
Kernel text base:  0xffffff80058e4000
__HIB  text base: 0xffffff8005700000
System model name: Macmini8,1 (Mac-7BA5B2DFE22DDD8C)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0'

> 'System uptime in nanoseconds: 13059054929
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000000030a616178
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000a55e1e46e 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff8044dc0000 - 0xffffffa044dc0000
  . PGZ   : 0xffffff8044dc0000 - 0xffffff8046dc1000
  . VM    : 0xffffff8046dc1000 - 0xffffff85135c0000
  . RO    : 0xffffff85135c0000 - 0xffffff86acdc0000
  . GEN0  : 0xffffff86acdc0000 - 0xffffff8b795c0000
  . GEN1  : 0xffffff8b795c0000 - 0xffffff9045dc0000
  . GEN2  : 0xffffff9045dc0000 - 0xffffff95125c0000
  . GEN3  : 0xffffff95125c0000 - 0xffffff99dedc0000
  . DATA  : 0xffffff99dedc0000 - 0xffffffa044dc0000
  Metadata: 0xffffffb05fe9e000 - 0xffffffb07fe9e000
  Bitmaps : 0xffffffb07fe9e000 - 0xffffffb082e9e000
  Extra   : 0 - 0
`

tenglendon, 2024-08-03:
testes

tenglendon, 2024-08-03:
tested
15.0 Beta 1 and 2 IntelBTPatcher.kext works well
15.0 Beta3 IntelBTPatcher.kext Kernel panik
15.0 Beta4 IntelBTPatcher.kext Kernel panik

SudoSuBash, 2024-08-07:
Yep.
The Intel BT Patcher couldn't work well on 15.0 Public Beta 3.
This is code:
`panic(cpu 2 caller 0xffffff801ff0b314): IOGMD: not wired for the IODMACommand @IOMemoryDescriptor.cpp:3248
Panicked task 0xffffff905870dd60: 201 threads: pid 0: kernel_task
Backtrace (CPU 2), panicked thread: 0xffffff9058723b30, Frame : Return Address
0xffffffa0cc5e7530 : 0xffffff801f75bd51 mach_kernel : _handle_debugger_trap + 0x4c1
0xffffffa0cc5e7580 : 0xffffff801f8c8b3c mach_kernel : _kdp_i386_trap + 0x11c
0xffffffa0cc5e75c0 : 0xffffff801f8b81bb mach_kernel : _kernel_trap + 0x48b
0xffffffa0cc5e7680 : 0xffffff801f6f1971 mach_kernel : _return_from_trap + 0xc1
0xffffffa0cc5e76a0 : 0xffffff801f75c047 mach_kernel : _DebuggerTrapWithState + 0x67
0xffffffa0cc5e77a0 : 0xffffff801f75b6e2 mach_kernel : _panic_trap_to_debugger + 0x1e2
0xffffffa0cc5e7810 : 0xffffff801ffd37d8 mach_kernel : _panic + 0x81
0xffffffa0cc5e7900 : 0xffffff801ff0b314 mach_kernel : __ZNK25IOGeneralMemoryDescriptor19dmaCommandOperationEjPvj + 0x894
0xffffffa0cc5e7970 : 0xffffff801ff0b5e1 mach_kernel : __ZN25IOGeneralMemoryDescriptor18getPhysicalSegmentEyPyj + 0x291
0xffffffa0cc5e7a50 : 0xffffff801ff0731c mach_kernel : __ZN18IOMemoryDescriptor9readBytesEyPvy + 0xfc
0xffffffa0cc5e7ab0 : 0xffffff802365992e com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x55
0xffffffa0cc5e7b00 : 0xffffff80220663e9 com.apple.iokit.IOUSBHostFamily : __ZN17AppleUSBIORequest8completeEv + 0xff1
0xffffffa0cc5e7cf0 : 0xffffff802117ad3b com.apple.driver.usb.AppleUSBCommon : __ZN24AppleUSBRequestCompleter20completeRequestQueueEP11queue_entry + 0x469
0xffffffa0cc5e7ef0 : 0xffffff802117a8b4 com.apple.driver.usb.AppleUSBCommon : __ZN24AppleUSBRequestCompleter12checkForWorkEv + 0x132
0xffffffa0cc5e7f20 : 0xffffff801fef6b0e mach_kernel : __ZN10IOWorkLoop15runEventSourcesEv + 0x13e
0xffffffa0cc5e7f60 : 0xffffff801fef6146 mach_kernel : __ZN10IOWorkLoop10threadMainEv + 0x36
0xffffffa0cc5e7fa0 : 0xffffff801f6f119e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.driver.usb.AppleUSBCommon(1.0)[066E3BA0-D5FC-3355-A355-BD6E44833135]@0xffffff8021178000->0xffffff802117bfff
         com.apple.iokit.IOUSBHostFamily(1.2)[A7025865-1246-3750-8AC9-9F9410682BCC]@0xffffff8022014000->0xffffff8022088fff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[27641E32-63A0-37ED-B1F7-326703B80BA4]@0xffffff8020b51000->0xffffff8020b54fff
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[A0274F23-311D-30DC-8C78-0BF4109AD475]@0xffffff8020e67000->0xffffff8020ea0fff
            dependency: com.apple.driver.AppleSMC(3.1.9)[4364B20F-0128-324C-AC42-FC3382C9784A]@0xffffff8020f59000->0xffffff8020f72fff
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[066E3BA0-D5FC-3355-A355-BD6E44833135]@0xffffff8021178000->0xffffff802117bfff
            dependency: com.apple.driver.AppleUSBHostMergeProperties(1.2)[E25ABE12-F5BF-36EC-B326-EF93B6C57C40]@0xffffff80220e7000->0xffffff80220e7fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[5CCDA23F-974B-36FE-991B-F492F97C7C90]@0xffffff802177c000->0xffffff802177dfff
         com.zxystd.IntelBTPatcher(2.5)[B4EA6DBC-E3C8-317A-B35F-6907682E6485]@0xffffff8023658000->0xffffff8023661fff
            dependency: as.vit9696.Lilu(1.6.8)[A4396287-D274-3C4A-B816-67B9F9022E01]@0xffffff80232c6000->0xffffff802334dfff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[A7025865-1246-3750-8AC9-9F9410682BCC]@0xffffff8022014000->0xffffff8022088fff

Process name corresponding to current thread (0xffffff9058723b30): kernel_task
Boot args: revpatch=sbvmm keepsyms=1 agdpmod=pikera debug=0x100 -revbeta -alcbeta

Mac OS version:
24A5309e

Kernel version:
Darwin Kernel Version 24.0.0: Wed Jul 31 21:46:13 PDT 2024; root:xnu-11215.0.199.501.2~1\/RELEASE_X86_64
Kernel UUID: 881546EF-0391-38D6-8C65-E3C0F28EA3BB
roots installed: 0
KernelCache slide: 0x000000001f400000
KernelCache base:  0xffffff801f600000
Kernel slide:      0x000000001f4e4000
Kernel text base:  0xffffff801f6e4000
__HIB  text base: 0xffffff801f500000
System model name: iMacPro1,1 (Mac-7BA5B2D9E42DDD94)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 18602640201
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x0000000454cdc5e5
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000001c9de283d1 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff80576d3000 - 0xffffffa0576d3000
  . PGZ   : 0xffffff80576d3000 - 0xffffff80596d4000
  . VM    : 0xffffff80596d4000 - 0xffffff8525ed3000
  . RO    : 0xffffff8525ed3000 - 0xffffff86bf6d3000
  . GEN0  : 0xffffff86bf6d3000 - 0xffffff8b8bed3000
  . GEN1  : 0xffffff8b8bed3000 - 0xffffff90586d3000
  . GEN2  : 0xffffff90586d3000 - 0xffffff9524ed3000
  . GEN3  : 0xffffff9524ed3000 - 0xffffff99f16d3000
  . DATA  : 0xffffff99f16d3000 - 0xffffffa0576d3000
  Metadata: 0xffffffa0a0dfb000 - 0xffffffa0c0dfb000
  Bitmaps : 0xffffffa0c0dfb000 - 0xffffffa0c3dfb000
  Extra   : 0 - 0
`

Daasin, 2024-08-15:
Seems to be on specific devices, works for me on Intel BE200 and AX210 (I think that was the 6E one anyways) both vPro variants.

The older one (also cnvi) seems to have the issue.

captain-spaghetti, 2024-08-29:
> Use this https://github.com/Lorys89/IntelBluetoothFirmware/actions/runs/9553221087/artifacts/1609892363

Do you know why my Logitech Craft keyboard will not connect?

I used your kexts to get bluetooth working again on the 15.1 dev beta, and everything else connects okay - but not the craft keyboard. I saw some posts a while back about the MX series of keyboards having issues with the paring step, but didn't seem to help me. I'm wondering if the Craft has it's own circumstances

tenglendon, 2024-09-08:
> tested 15.0 Beta 1 and 2 IntelBTPatcher.kext works well 15.0 Beta3 IntelBTPatcher.kext Kernel panik 15.0 Beta4 IntelBTPatcher.kext Kernel panik

15.0 Beta3～8 IntelBTPatcher.kext Kernel panik
有其他解決辦法嗎？
還是得乖乖等作者更新？

ecklf, 2024-09-17:
Fix that works on my Intel AX-210 (non vPro) on 15.0 stable
Kexts and diff of `config.plist` included in zip
[image]
[BT_SEQUOIA.zip](https://github.com/user-attachments/files/17029925/BT_SEQUOIA.zip)

Lorys89, 2024-09-17:
> ```shell
> 
> com.apple.driver.IOPlatformPluginLegacy	1.
> ```

work for me :)
and the source code?

Hazer, 2024-09-17:
Works for me too! Thanks :)

zxystd, 2024-09-17:
[IntelBTPatcher.kext.zip](https://github.com/user-attachments/files/17036455/IntelBTPatcher.kext.zip)
This is a pre-release one, please help to check.

Hazer, 2024-09-18:
> [IntelBTPatcher.kext.zip](https://github.com/user-attachments/files/17036455/IntelBTPatcher.kext.zip) This is a pre-release one, please help to check.

@zxystd just to make sure I'll test this the right way:

1. I should use which IntelBluetoothFirmware and BlueToolFixUp kexts? Latest github action versions?
2. I should use nvram flags bluetoothExternalDongleFailed, bluetoothInternalControllerInfo with this version too?

oukb, 2024-09-19:
AX211 works, but doesn't find devices

tenglendon, 2024-09-19:
1.KILLER 1550 = 1960AC works,but can't find logitech LE kind mouse
2. can use APPLE magic mouse 1 and 2
3. I used to use [Lorys89](https://github.com/Lorys89) fork and logitech LE kind mouse work ok,but now his fork
[IntelBTPather](https://github.com/Lorys89/IntelBluetoothFirmware/actions/runs/10915759387) still cause panic
4.flags 
bluetoothExternalDongleFailed DATA 00
bluetoothInternalControllerInfo DATA 0000000000000000000000000000
is meaningful and nessary
but dealt makes no difference

Takudzwamz, 2024-09-20:
> Fix that works on my Intel AX-210 (non vPro) on 15.0 stable Kexts and diff of `config.plist` included in zip. Ensure to clean NVRAM. ![CleanShot 2024-09-17 at 16 34 57@2x](https://private-user-images.githubusercontent.com/8146736/368211795-71b76a52-0f6a-4d56-8c36-2ce4e75fdf95.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY4MTA3MDUsIm5iZiI6MTcyNjgxMDQwNSwicGF0aCI6Ii84MTQ2NzM2LzM2ODIxMTc5NS03MWI3NmE1Mi0wZjZhLTRkNTYtOGMzNi0yY2U0ZTc1ZmRmOTUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MjBUMDUzMzI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTIxMTk4MTkzMmU1YWZhYjViNDBmMjEyYzA5NTQ2MzQ1YWMyMDg2Njc4MDhlMGYzNjhiMjg5MmFmZjU4OGU5MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.TkWrt-IYJlbveHdmP4ybV1xT_xDzZgOuOBp3v4oUqWc) [BT_SEQUOIA.zip](https://github.com/user-attachments/files/17029925/BT_SEQUOIA.zip)

This worked for me on Sequoia's publicly available version 15.0. My Bluetooth now works.

ngophuong, 2024-09-20:
> Fix that works on my Intel AX-210 (non vPro) on 15.0 stable
> 
> Kexts and diff of `config.plist` included in zip. Ensure to clean NVRAM.
> 
> [image]
> 
> [BT_SEQUOIA.zip](https://github.com/user-attachments/files/17029925/BT_SEQUOIA.zip)
> 
> 

This one work for me as well when i upgrade from sonoma to sequoia from official release [full download]

zxystd, 2024-09-20:
> Works for me too! Thanks :)

just replace IntelBTPatcher.kext, others keep the same as previous working.

naidb, 2024-09-21:
> > Works for me too! Thanks :)
> 
> just replace IntelBTPatcher.kext, others keep the same as previous working.
It worked for me
[https://www.insanelymac.com/forum/topic/359997-pre-release-macos-sequoia-151/?do=findComment&comment=2825662](url)Clover
[https://www.insanelymac.com/forum/topic/359997-pre-release-macos-sequoia-151/?do=findComment&comment=2825652](url)Opencore

Sequoia Sonoma ventura monterey bigsur >>>2.5.0
thanks. you verymuch

carlosaclima, 2024-09-24:
> This worked for me on Sequoia's publicly available version 15.0. My Bluetooth now works.
It worked for me! Success! Thanks for everyone's effort.
[image]

Hazer, 2024-09-25:
> [IntelBTPatcher.kext.zip](https://github.com/user-attachments/files/17036455/IntelBTPatcher.kext.zip) This is a pre-release one, please help to check.

It's also working for me! Thanks :)

VGerris, 2024-09-26:
> @Lorys89 , The kexts you posted work perfectly. Thank you very much for that.
>
> I don't need this: bluetoothExternalDongleFailed DATA 00 bluetoothInternalControllerInfo DATA [...]

I needed -btlfxallowanyaddr to make it work on my Dell Precision 5550. without it bluetooth would not work.
With it it works on Sequoia 15.0 release, thank you

raczalex, 2024-10-18:
Hi do you guys know any kexts or fixes for Logitech BT LE devices?
I'm have sequioa 15.0.1 installed
bluetooth works i just cannot connect logitech mx keys mini keyboard

[image]

jzrodriguez98, 2024-10-18:
Was the FBPatcher kext supposed to be used for LE BT devices? Are you using it?

jzrodriguez98, 2024-10-18:
> Hi do you guys know any kexts or fixes for Logitech BT LE devices?
> 
> I have sequioa 15.0.1 installed
> 
> bluetooth works i just cannot connect logitech mx keys mini keyboard
> 
> 
> 
> [image]
> 
> 

Was the IntelBTPatcher kext supposed to be used for LE BT devices? Are you using it?

raczalex, 2024-10-19:
Yes im using it.

jzrodriguez98, 2024-10-20:
> Yes im using it.

Got it. Did you compile the latest version of the kexts available in github?

raczalex, 2024-10-21:
i tried 2.5.0 release and other versions too, i cannot compile all the kexts from Lorys89 because i got an error message with intelBTPatcher kext

jzrodriguez98, 2024-10-22:
> i tried 2.5.0 release and other versions too, i cannot compile all the kexts from Lorys89 because i got an error message with intelBTPatcher kext

I compile the original kexts using a very helpful tool created by that allows us to compile all Intel wifi and bluetooth kexts.

chris1111, 2024-10-22:
> > i tried 2.5.0 release and other versions too, i cannot compile all the kexts from Lorys89 because i got an error message with intelBTPatcher kext
> 
> I compile the original kexts using a very helpful tool created by @chris1111 that allows users to compile all Intel wifi and bluetooth kexts (https://github.com/chris1111/Wifi-Intel-KextsBuilder). For @Lorys89 forks I have used the release versions.

Hi @jzrodriguez98 
Here a version Build all itlwm with Lorys89/IntelBluetoothFirmware
[Wifi-Intel-KextsBuilder.zip](https://github.com/user-attachments/files/17468485/Wifi-Intel-KextsBuilder.zip)

I just change the link in Line 65

[image]

chris1111, 2024-10-23:
@jzrodriguez98  @raczalex 
Here attaching a version can build all from @Lorys89  Repos
[Wifi-Intel-KextsBuilder.zip](https://github.com/user-attachments/files/17483904/Wifi-Intel-KextsBuilder.zip)

[image]

fabianaltensleben, 2024-10-26:
kann ich leider nicht verwenden da ich keinen Dev Account habe bei Apple . 
Könntet Ihr bitte die Kompilierten Kext online stellen ? Danke

kalkmann, 2024-12-06:
> Hi do you guys know any kexts or fixes for Logitech BT LE devices? I have sequioa 15.0.1 installed bluetooth works i just cannot connect logitech mx keys mini keyboard
> 
> ![Screenshot 2024-10-18 at 16 55 16](https://private-user-images.githubusercontent.com/126008971/377915056-2051818f-21a0-4c18-ae4c-af1967f03bb2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM0ODg5NTQsIm5iZiI6MTczMzQ4ODY1NCwicGF0aCI6Ii8xMjYwMDg5NzEvMzc3OTE1MDU2LTIwNTE4MThmLTIxYTAtNGMxOC1hZTRjLWFmMTk2N2YwM2JiMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwNlQxMjM3MzRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZTQ1ZjdjZDc2ZWJiMTZkZTUyZWZhMGQ5ZTlkMjhhMzY3ZDQwOTZkNTYyOGZmMTQxNjJiNDE3YTlhNDhmMDUwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.x7KXb0PUfmye7X9X2LynTUuclm01qbIgYeMydwjlBL8)
Have you had any luck connecting the Logitech? I'm having trouble connecting an MX Master 3S mouse. I'm using Sequoia 15.1.1 and my Intel is AX210. The mouse is visible on the Bluetooth settings screen, but it doesn't connect (timeout). I tested it with another mouse (Xiaomi Mi Silent) and the same behavior occurs.

I tested it by creating the Kexts from @Lorys89's repo (with the help of the Wifi-Intel-KextsBuilder app from @chris1111), but I get the Kernel Panic from the post https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/486#issuecomment-2273609683

If anyone is successfully using a BT LE mouse, I'd appreciate any information on how to proceed.

my05, 2024-12-16:
Tried all of the suggestions above on Sequioa 15.1 (AX210 hack) but none of them works. When trying chris1111 from the 22nd of October and replacing the kexts, i got into this ever running boot-loop. Other kexts above simply did not work at all unfortunately (still can see the BT mouse or even mice, just can't pair them).

Whiskey Lake I5 (Fujitsu)

jzrodriguez98, 2024-12-17:
> Tried all of the suggestions above on Sequioa 15.1 (AX210 hack) but none of them works. When trying chris1111 from the 22nd of October and replacing the kexts, i got into this ever running [...]

Have you tried the nvram fix below?

sudo NVRAM bluetoothExternalDongleFailed=%00
sudo pkill bluetoothd

you just need to check what's going on in nvram using 'NVRAM -p’ in terminal and if you have  bluetoothExternalDongleFailed=%01 up there, use the commands above.

plooshi, 2024-12-19:
https://github.com/Ploosho/IntelBluetoothFirmware/actions/runs/12417106086/artifacts/2344402393 this should be able to do LE on sequoia

jzrodriguez98, 2024-12-20:
> https://github.com/Ploosho/IntelBluetoothFirmware/actions/runs/12417106086/artifacts/2344402393 this should be able to do LE on sequoia

I tried to open this link but it didn't work for me

kalkmann, 2024-12-20:
> https://github.com/Ploosho/IntelBluetoothFirmware/actions/runs/12417106086/artifacts/2344402393 this should be able to do LE on sequoia

[image]

Thank you!!!

plooshi, 2024-12-20:
> > https://github.com/Ploosho/IntelBluetoothFirmware/actions/runs/12417106086/artifacts/2344402393 this should be able to do LE on sequoia
> 
> I tried to open this link but it didn't work for me
> 
> @plooshi Is this run linked here the one that you shared

what do you mean
if you're asking for the run link, https://github.com/Ploosho/IntelBluetoothFirmware/actions/runs/12417106086

tenglendon, 2024-12-20:
> > > https://github.com/Ploosho/IntelBluetoothFirmware/actions/runs/12417106086

with this 2 Kext,my Master 3S and logitech Craft Keyboard bluetooth connect like a charm~

thanks to much~

plooshi, 2024-12-20:
can someone test the new artifacts

kalkmann, 2024-12-20:
> can someone test the new artifacts

Deleted repo?

plooshi, 2024-12-20:
> > can someone test the new artifacts
> 
> Deleted repo?

moved to personal repo as i use orgs for stuff that are either experiments or aren't complete

[63 more replies]