OpenIntelWireless/IntelBluetoothFirmware #490: 蓝牙无法使用

akwangl, 2024-06-24:
### Have you read the docs?

Yes

### macOS Version

macOS 12.7.5

### Kext Version

v2.4.0

### Wireless Adapter Model and USB Product ID

AX200

### Description

## EFI
[image]

当睡眠后就无法开启蓝牙，有时候打开控制面板还会报无法加载蓝牙设备

### Info in System Report - Bluetooth

```shell
蓝牙控制器：
  地址：	NULL
  状态：	关闭
  芯片组：	BCM_4350C2
  可被发现：	关闭
  固件版本：	v0 c0
  产品ID：	0x0001
  支持的服务：	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
  传输：	USB
  供应商ID：	0x004C (Apple)
```

### Relevant log output

```shell
➜  ~ sudo dmesg | grep -i IntelBluetooth
Password:
➜  ~ sudo dmesg | grep -i IntelBluetooth
➜  ~ sudo dmesg | grep -i IntelBluetooth
➜  ~ sudo dmesg | grep -i IntelBluetooth
➜  ~
```

akwangl, 2024-06-24:
调整了一下efi顺序现在正常了

akwangl, 2024-06-24:
# 之前
[image]
# 之后
[image]

akwangl, 2024-06-26:
[image]

当休眠一段时间后，会偶发性从ax200变成BCM_4350C2。这种情况只能重启解决

zxystd, 2024-06-26:
Hackintosh hibernation issue, fix it by yourself.

closed 2024-06-26 by zxystd

Jackhr-arch, 2024-06-26:
请问如何排序？我当前也是处于BCM状态，对你发布的efi做了一些修改

akwangl, 2024-07-02:
> 请问如何排序？我当前也是处于BCM状态，对你发布的efi做了一些修改

把睡眠关了，一直稳定