OpenIntelWireless/IntelBluetoothFirmware #492: USB ports broken after the commit 3e5c1e60269ae0329094de131227285d4682b665

z3DD3r, 2024-07-23:
### Have you read the docs?

Yes

### macOS Version

macOS 14.5

### Kext Version

v2.5.0

### Wireless Adapter Model and USB Product ID

Intel AX210

### Description

USB ports doesn't work after the commit 3e5c1e60269ae0329094de131227285d4682b665 on macOS 14.5
I've compiled IntelBTPatcher myself from master branch and noticed that my iPhone can not be connected to my hackintosh. Nothing happens after connection. Literally nothing. Even charging doesn't start. After reverting this commit and recompiling kext everything starts work without any issues...

### Info in System Report - Bluetooth

```shell
-
```

### Relevant log output

```shell
-
```

zxystd, 2024-08-19:
Thank you for the report, is there anyone else can also confirm?

zxystd, 2024-09-17:
Can you use the latest one to check again? if not work, please use this one. 
[IntelBTPatcher.kext.zip](https://github.com/user-attachments/files/17036453/IntelBTPatcher.kext.zip)

z3DD3r, 2024-09-18:
@zxystd
I've checked the 01cc1806d71f5cc64c464851b9f4811a3e7b4791 commit. Everything works fine for me on MacOS 14.6. Thanks a lot 👍

zxystd, 2024-09-18:
ok, thank you. you are sure 50e971c524081a1d97a72794db2640aa0136e284 this commit broke the usb?

z3DD3r, 2024-09-18:
> ok, thank you. you are sure [50e971c](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/50e971c524081a1d97a72794db2640aa0136e284) this commit broke the usb?

Yes. I've tested it once again on MacOS 14.6. Commit 50e971c524081a1d97a72794db2640aa0136e284 brokes usb.

closed 2024-09-20 by z3DD3r