OpenIntelWireless/IntelBluetoothFirmware #494: Bluetooth os not working

SamuelPalubaCZ, 2024-08-26:
### Have you read the docs?

Yes

### macOS Version

somala

### Kext Version

latest 

### Wireless Adapter Model and USB Product ID

Intel DualBand WiFi Card 00JT464

### Description

im just stupid or idk, I turned on bluetooth at settings and it "started" bluetooth but show no devices, when I reopen settings it appears to be off

### Info in System Report - Bluetooth

```shell
Bluetooth řadič:
  Adresa:	NULL
  Stav:	Vypnuto
  Čipová sada:	BCM_4350C2
  Zjistitelné:	Vypnuto
  Verze firmwaru:	v0 c0
  Podporované služby:	0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  Přenos:	USB
  ID dodavatele:	0x004C (Apple)
```

### Relevant log output

```shell
Last login: Tue Aug 27 00:24:54 on console
^[[A%                                                                           ➜  ~ sudo dmesg | grep -i IntelBluetooth
Password:
[    2.096347]: IOUSBHostDevice@14400000: IOUSBHostDevice::setConfigurationGated: IntelBluetoothFirmware selected configuration 1
```

oxyii, 2024-09-20:
> somala

I'm not sure that it is correct. Maybe Sequoia? Or Sonoma? Anyway please check installed versions of Lilu.kext and BlueToolFixup.kext

I had a similar problem on macOS Sequoia (15.0). In my case it was "Disabled BlueToolFixup plugin" because of incompatible os version ([source](https://github.com/acidanthera/Lilu/blob/master/Lilu/Sources/kern_api.cpp#L91)).

If your os is Sequoia please see this [PR](https://github.com/acidanthera/BrcmPatchRAM/pull/33) especially [this comment](https://github.com/acidanthera/BrcmPatchRAM/pull/33#issuecomment-2357998889). Replacing BlueToolFixup.kext with the one in the comment helped me.

MatteoBax, 2024-09-29:
Bluetooth doesn't work for me either.

> I had a similar problem on macOS Sequoia (15.0). In my case it was "Disabled BlueToolFixup plugin" because of incompatible os version [...]

I tried what you said on macOS Sequoia but Bluetooth still doesn't work
My network card is Intel Wireless-AC 9560.

jzrodriguez98, 2024-09-29:
@MatteoBax you should try following this guide [](https://elitemacx86.com/threads/how-to-fix-intel-wifi-and-bluetooth-on-macos-sequoia-and-later.2066/).