OpenIntelWireless/IntelBluetoothFirmware #497: Keyboard won't pair and no handoff in Sonoma

emulatorchen, 2024-11-09:
### Have you read the docs?

Yes

### macOS Version

macOS 14.7.1

### Kext Version

v2.4.0

### Wireless Adapter Model and USB Product ID

Intel AX201

### Description

I can have below devices working

- Logitech MX Ergo 
- Apple Trackpad 2 
- AirPods Pro

Working started from 14.2 to 14.7.1 with release version 2.4.0 and other potential test versions in those open discussions but never had a chance to get my keyboard working. These devices did show in the list when entering pairing mode but there's no pin code dialog show up so it never succeeds.

-  Logitech 913TKL
- Realforce R3 
- Filco covertiable 2

These keyboards are working fine with USB and bluetooth under Windows.

Kexts in OC folder:
ls /Volumes/EFI/EFI/OC/Kexts
AirportItlwm.kext/           CPUFriend.kext/              IntelMausi.kext/             SMCRadeonGPU.kext/
AppleALC.kext/               CPUFriendDataProvider.kext/  Lilu.kext/                   SMCSuperIO.kext/
AppleIGC.kext/               DebugEnhancer.kext/          RadeonSensor.kext/           USBPorts.kext/
BlueToolFixup.kext/          IntelBTPatcher.kext/         RestrictEvents.kext/         VirtualSMC.kext/
BrcmPatchRAM3.kext/          IntelBluetoothFirmware.kext/ SMCProcessor.kext/           WhateverGreen.kext/

[image]

### Info in System Report - Bluetooth

```shell
Bluetoothコントローラ:
  アドレス:	04:EC:D8:E0:9C:40
  状態:	オン
  チップセット:	THIRD_PARTY_DONGLE
  検出可能:	オフ
  ファームウェアのバージョン:	v13719 c13719
  対応サービス:	0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  転送:	USB
  製造元ID:	0x004C (Apple)
  接続済み:
  Emulator のトラックパッド:
  アドレス:	6C:AB:31:F3:67:3C
  製造元ID:	0x004C
  製品ID:	0x0265
  ファームウェアのバージョン:	3.1.1
  種類（マイナー）:	AppleTrackpad
  サービス:	0x800020 < HID ACL >
```

### Relevant log output

```shell
sudo dmesg | grep -i IntelBluetooth
Password:
[    6.566467]: IOUSBHostDevice@14e00000: IOUSBHostDevice::setConfigurationGated: IntelBluetoothFirmware selected configuration 1
```

schrup21, 2025-01-24:
Did you already try -btlfxallowanyaddr in boot-args?

My Logitech MX Mouse couldn't get paired (was found but nothing happened), I tried pairing several times.
After adding -btlfxallowanyaddr unter NVRAM 7C436110-AB2A-4BBB-A880-FE41995C9F82 boot-args, the issue was gone - but maybe that was just a coincidence and it just needed more attempts..

emulatorchen, 2025-01-30:
Yes I already did and tried it with several debug versions. I also found that I could not pair with my 3-party switch controller, similar situation like keyboard, was found but not paired. But the game controller should not require a pair code, so not sure what's going on.

[image]

emulatorchen, 2026-10-10:
Solved by https://github.com/OpenIntelWireless/IntelBluetoothFirmware/pull/515

closed 2026-11-08 by emulatorchen