OpenIntelWireless/IntelBluetoothFirmware #501: BT not patches on macOS Sonoma

Ampernic, 2025-05-26:
### Have you read the docs?

Yes

### macOS Version

macOS 14.7.6

### Kext Version

v2.4.0 - v2.5.0 ( 01cc180 ) 

### Wireless Adapter Model and USB Product ID

Intel Centrino Advanced-N 6235 ( 0x07da )

### Description

[image]

(BT shown in USB section of system report correctly, and setup as builtin)
[image]

USB Map works correct

[image]

But not working 

[image]

Kexts order 

[image]

Trying also sequoia nvram entries (with and without)

[image]

### Info in System Report - Bluetooth

```shell
Контроллер Bluetooth:
  Адрес:	NULL
  Состояние:	Выкл.
  Набор микросхем:	BCM_4350C2
  Видимо для других устройств:	Выкл.
  Версия прошивки:	v0
  Поддерживаемые службы:	0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  Передача данных:	PCIe
  ID поставщика:	0x004C (Apple)
```

### Relevant log output

```shell
ampernic@iMac-Ampernic ~ % sudo dmesg | grep IntelFirmware
ampernic@iMac-Ampernic ~ % 

(yeah, nothing...)
```

### Kernel Panic Logs

```shell

```

fidelio959, 2025-06-02:
Hello I updated to Sonoma 14.7.6 and I lost bluetooth. I tried your solution injecting brcmpatches with boot argument but still no work. Can you tell me which bluetooth kext are you using? I got the same chipset on an Asus b360 board.

amane1234, 2025-06-17:
I don't know the reason why we need BrcmPatch for intelBT hardware, but the absence of BrcmPatch makes wifi speed significantly slow.

Solution : Enabled VT-D with DMAR.aml patch + intelbluetooth 2.5.0 (IntelBTPatcher + IntelBueltoothFirmware) + BrcmPatch with boot-args (-btlfxallowanyaddr-btlfxboardid) make my two intelbluetooth (AX201, 8260) stable, improving wifi speed.

Thanks Ampernic.