OpenIntelWireless/IntelBluetoothFirmware #507: Project is dead?

SerjNIX, 2025-09-26:
### Have you read the docs?

Yes

### macOS Version

26

### Kext Version

2.4.0

### Wireless Adapter Model and USB Product ID

ax211 0x0033

### Description

- no merged pull requests
- no updates
- no replies in issues
R.I.P.

### Info in System Report - Bluetooth

```shell
Bluetooth USB Host Controller:

  Location ID:	0x14a00000
  Connection Type:	Built-in
  Serial Number:	Not Provided
  Link Speed:	12 Mb/s
  USB Vendor ID:	0x8087
  USB Product ID:	0x0033
  USB Product Version:	0x0000
  Power Allocated:	0.5 W (100 mA)
```

### Relevant log output

```shell
empty
```

### Kernel Panic Logs

```shell
no panic
```

davidtran1010, 2025-09-28:
I have the same issue, bluetooth of AX210 not working now, I have tried many ways

SerjNIX, 2025-10-27:
> any news?

Hi, I use solution from here https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/505#issuecomment-3370000263

There are no reaction in this project by maintainers.

MinecraftNerrd1337, 2026-10-09:
@SerjNIX 
Does the fork fix AX210 for older versions of macOS as well? I'm interested because it's been broken for quite a while, I have reported here: https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/496
I've even downgraded one of my hackintoshes to an AX200, as it's AMD based and using the older version of the kexts (which was my fix) causes random KPs.